robotology/icub-main #952: Improve the status handler for BAT and BMS

valegagge, 2024-03-18:
### Is your feature request related to a problem?

Improve the status information related to the BMS and BAT. Currently, both objects rely on the `embObjBattery` device which is not at all compliant with the BAT board.

Even if the best solution is to create an ad-hoc device and expose ad-hoc API in a new interface, I think that the first step is to improve the returned status. 
Currently, the status is not related to the enum exposed in the API interface but a number that is a bitmask of the status related to BAT or BMS. In this way the user cannot use the status the information.

 

### The solution you would like to have available

The idea here is to map the status of BAT and BMS to a significant value in the enum provided by the API and update the documentation accordingly. 

[image]

valegagge, 2024-03-18:
cc @MSECode

pattacini, 2024-03-26:
Hi @MSECode @valegagge 

Is this related to #953?

MSECode, 2024-03-26:
Not really, it is related to further improvements we wanna port to the management of the status message for both BAT and BMS and their relation with the yarp interface iBattery.

valegagge, 2024-03-27:
We are waiting for the new BMS to finalize the work.