robotology/icub-main #959: Fix the motor temperature spike filtering  in embObjMotionControl

valegagge, 2024-05-07:
### Bug description

In embObjMotionControl, a temperature monitor has been implemented to provide a warning to the user when the temperature goes beyond the warning threshold defined in the configuration file.
Since the motor temperature reading suffers from disturbance there are some spikes that have not be take in consideration in the check of the warning threshold.

In addition, the current fw sends some negative values to encode some error cases. Also, this should be removed from the warning check.

### Steps to reproduce

Just read the motor temperature. (In some motors the readings are noisier than others)

### Expected behavior

In the log, there are any warnings related to the spikes of temperature

valegagge, 2024-05-07:
Just a tip: for debugging and making the work, you can create an ad hoc fw that sometimes sends a spike.
Of course, the final test has to be done on the robot

MSECode, 2024-05-15:
After developing a first version of the filtering in the sw I've tested it by generating synthetic data with the aim of simulating a continuous rising of the temperatures with a pyramidal pattern. Basically, starting from 0 degree I've then risen the temperature of about 10 degree, so that the threshold between temperature at time interval `i` and `i-1` (20 degree) it is not overcome. Then, after some steps I've instead increment the temperature to a higher value (around 110/120 degree in order to simulate a spike). And finally I've decreased again the temperature value.
Part of the described behavior is shown in this image

[image]

Doing so, and looking at the logs of the YRI, it is possible to see that we do not have any warning related to the generated spike but the warning are just related to the overcoming of the warning temperature limit (set now to 40 degree just for test purposes) when the delta between current and previous temperature is lower than a defined threshold.

[image]

cc: @valegagge

MSECode, 2024-05-15:
/reminder May, 21, 2024

@valegagge @MSECode

MSECode, 2024-05-21:
## WIP
 my branch 

[image]

MSECode, 2024-05-22:
PR opened here:

https://github.com/robotology/icub-main/pull/961

pattacini, 2024-05-24:
PR merged ✅ 
Probably, this issue can be closed as complete.

cc @MSECode @valegagge

closed 2024-05-24 by valegagge