robotology/icub-main #966: yarprobotinterface with faceExpression

AlessandroT99, 2024-05-31:
### Bug description

We updated the iCubGenova07 to the latest release of robotology (2024.02.0), and we noticed that the faceExpression module changed its startup with respect to our old version. So from using yarpdev, now it needs yarprobotinterface.

Up to know we have not find a proper way to both startup iCub and the faceExpression module.

### Steps to reproduce

in one terminal:
- yarprobotinterface

in another:
- yarprobotinterface --context faceExpressions --config serialPortFaceExpressions.xml

### Expected behavior

I expect to get a address conflit from the step above, and so it is.
The problem is, how can I do both startup properly?

### Additional context

Up to know we have found a temporary solution which consist of:
- adding at the end of icub_all.xml configuration file the following line `<xi:include href="../../../iCub/contexts/faceExpressions/serialPortFaceExpressions.xml" />`
- commenting the line `<robot name="iCubFaceExpressions" build="2" portprefix="/icub" xmlns:xi="http://www.w3.org/2001/XInclude">` into serialPortFaceExpression.xml

Then execute simply yarprobotinterface.
This seems not a proper way to do it, have you any suggestion?

Thank you so much!

pattacini, 2024-05-31:
Hi @AlessandroT99 

`yarprobotinterface` (aka YRI) is simply a launcher, or a deployer if you will. Thus, it serves to start a bunch of services, depending on the command line options you pass in.

In light of this, the following statement is not generally true:

>I expect to get a address conflit from the step above, and so it is.

Two instances of YRI can be perfectly run side by side, guaranteeing that they're properly configured to launch different services, as it happens for starting the robot and the face expressions.

Therefore, everything boils down to configuring the two instances.
In particular, the first instance refers to a default behavior since there ain't command line options.

To find out what conf file is retrieved by the first instance, try executing:
```console
yarp resource --from yarprobotinterface.ini
```
The outcome depends on many factors as reported in:
- https://github.com/orgs/robotology/discussions/42

AlessandroT99, 2024-05-31:
Hi @pattacini,

Thank you for your answer! 
The default behavior of yarprobotinterface in our case is activating icub_all.xml in our robot folder (build/install/.../iCubGenova07/yarprobotinterface.ini).

However, after failing for the whole day, now everything works following the order of the commands reported before.

We have no idea of what we were missing or doing wrong.

Anyway thank you so much for your help!

Alessandro

closed 2024-05-31 by AlessandroT99