robotology/icub-main #973: Remove the implementation of `yarp::dev::IAnalogSensors` from the `embObjPOS` device

valegagge, 2024-07-23:
### Is your feature request related to a problem?

Please see the issue #929 

### The solution you would like to have available

Please see this comment https://github.com/robotology/icub-main/issues/929#issuecomment-2194116489

valegagge, 2025-04-15:
We can substitute the IAnalog interface with [IEncoderArrays](https://www.yarp.it/latest/classyarp_1_1dev_1_1IEncoderArrays.html). we can follow the axample of 'embObjMAIS`

MSECode, 2025-05-02:
I've started to substitute the `IAnalogSensor` interface with `IEncoderArrays`. At the moment and according to the current code we have in `embObjPOS` https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjPOS/embObjPOS.cpp#L266-L317, I'm keeping the implementation of the new interface methods empty by just returning zero.
However, since I suppose we'd have the methods actually fully implemented, we should study how the new interface has been though. Specifically, I'm referring to how the encoder array should be intended to work. Basically, I'm wondering if either:
- an encoder array is referred to be correlated to the single joint, which then have multiple encoders per joint
- an encoder array is intended to be correlated to the `embObjPOS`object (i.e. I'm referring to how it is currently managed by the POS wrappers), , meaning that the array of encoder is made by either the 4 open-close joints of the hand or the 2 abduction joints of the hand. So, in this way the specific encoder array would be made by the encoder of multiple joints.
- both ways can be used for describing the system, thus we can define methods as we desire depending on the way we define the encoder array structure

MSECode, 2025-05-06:
In order to update the  `embObjPOS`  software to a stable condition, I'd like to understand how the `IEncoderArrays` interface has been thought. 
Probably @traversaro can help me on that. Thus, considering that:
- the POS service has been made to work with the FAP sensor, which in our case reads the position of the single finger joint
- the POS service configures the hand joints attaching the 4 open-close of one hand to one motion control device and the 2 abduction to another motion control device as you can see here: https://github.com/icub-tech-iit/robots-configuration/blob/master/ergoCubSN001/hardware/motorControl/left_arm-eb25-j11_12-mc_service.xml and https://github.com/icub-tech-iit/robots-configuration/blob/master/ergoCubSN001/hardware/motorControl/left_arm-eb23-j7_10-mc_service.xml
when overriding these interface methods: https://www.yarp.it/latest/classyarp_1_1dev_1_1IEncoderArrays.html, how should I thought them. Specifically, I'm wondering if in this case:
- the encoder array should be referred to each single joint and thus each array would have dimension 1 --> e.g. looking to this method: `getEncoderArraySize(size_t sens_index)`, `sens_index` would be the index of the specific finger (ideally from 0 to 6) 
- the encoder array should be referred to the "motion control device", thus if referring to the method above, the size of the array should be 2 for the abduction and 4 for the open close and the index would ideally go from 0 to 2
- either way can be used and it is up to us how we build the service.

traversaro, 2025-05-06:
I would just keep things simple and migrate a single `embObjPOS` device from exposing a single 2/4-sized vector from the `IAnalogSensors` interface to a IEncoderArrays that expose 1 encoderArray of size 2 or 4, as done in https://github.com/robotology/icub-main/pull/905 . We can understand if it make sense then to expose this info in some smarter way, but that seems a different concern w.r.t. to migrate out of `yarp::dev::IAnalogSensors`.

valegagge, 2025-05-12:
Hi @MSECode ,
I did a quick review on the code you changed as we agreed. I commit on your branch some suggestions that regarding some missing point we have already discussed f2f. Let me know if they are fine for you.

Thanks

valegagge, 2025-05-13:
After a f2f with@MSECode , the missing points were described in the .h file as comments. 
We just missing to understand how the fw fits the pos code in the array it sends.

MSECode, 2025-05-28:
I've checked on `ergoCub002` what we are streaming on the POS ports if we enable the POS wrappers for open-close and abduction. I've done this in order to check what the `read` method of the `POSService` is actually doing.
Thus, I've seen that independently to the joints that we have enabled, the POS service is always streaming an array of 14 values.
Then, POS4 port, which is related to the open-close, streams the 4 position of the 4 open-close in the first 4 indexes. 
Instead, POS2 port, which is related to the abduction, streams 2 position, which in this case does not match with what we see in the `motorgui`, at index 4 and 6. 

Here you can see what I've read from those port:
- left hand POS 4
`15.0 5.10000000000000053291 4.80000000000000071054 5.10000000000000053291 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0`
- left hand POS2
`0.0 0.0 0.0 0.0 360.600000000000022737 0.0 -32.8000000000000042633 0.0 0.0 0.0 0.0 0.0 0.0 0.0`

This help us in understanding how we can refactor and clean the read method of the POS service, even though the POS port are usually never opened on the robot

MSECode, 2025-06-06:
Sure, we should investigate better on that

MSECode, 2025-08-29:
After some tests on `ergoCub001` we have observed that the fingers FAP positions are actually sent from the boards to the embObj device as shown by the image below, where we have added a debug print in the `update` method in embObjPOS.

[image]

However, we have realized that the method that should be called for sending the data to the yarp port, i.e. `getEncoderArrayMeasurement`, is actually never called and this is the reason why we do not read any data in the bottle in the published port.