robotology/icub-main #977: http://wiki.icub.org/wiki/Manual website linked in the documentation is down?

traversaro, 2024-08-02:
### Bug description

The link  http://wiki.icub.org/wiki/Manual linked in https://robotology.github.io/robotology-documentation/doc/html/index.html (and also in other points of the repo) seems not to be working. 

It is probably not a big problem, as most of the material is now available in https://icub-tech-iit.github.io/documentation/, but if that is the case we should probably update the link.

### Steps to reproduce

.

### Expected behavior

.

### Example repository

.

### Additional context

.

pattacini, 2024-08-02:
Good point 👍🏻 
I'll take care of it.

traversaro, 2024-08-02:
For reference, the latest version stored in the Internet Archive is http://web.archive.org/web/20230223191757/http://wiki.icub.org/wiki/Manual .

closed 2024-08-02 by pattacini