robotology/icub-main #988: CI failure w/ SWIG on Python about `createVocab32`

pattacini, 2024-10-14:
### Bug description

As of today, the CI fails while Swig compiles icub.i for python:

```console
2024-10-14T06:09:47.4155353Z /home/runner/work/icub-main/icub-main/install/include/yarp/sig/Image.h:42: Error: Type error. Expecting an integral type
...
2024-10-14T06:09:47.4197493Z /home/runner/work/icub-main/icub-main/install/include/yarp/sig/Image.h:68: Error: Type error. Expecting an integral type
2024-10-14T06:09:47.4870606Z /home/runner/work/icub-main/icub-main/install/include/yarp/dev/IAxisInfo.h:24: Error: Type error. Expecting an integral type
2024-10-14T06:09:47.4873013Z /home/runner/work/icub-main/icub-main/install/include/yarp/dev/IAxisInfo.h:25: Error: Type error. Expecting an integral type
2024-10-14T06:09:47.4875108Z /home/runner/work/icub-main/icub-main/install/include/yarp/dev/IAxisInfo.h:27: Error: Type error. Expecting an integral type
2024-10-14T06:09:47.5572523Z /home/runner/work/icub-main/icub-main/install/include/yarp/dev/IInteractionMode.h:18: Error: Type error. Expecting an integral type
2024-10-14T06:09:47.5574884Z /home/runner/work/icub-main/icub-main/install/include/yarp/dev/IInteractionMode.h:19: Error: Type error. Expecting an integral type
2024-10-14T06:09:47.5577180Z /home/runner/work/icub-main/icub-main/install/include/yarp/dev/IInteractionMode.h:21: Error: Type error. Expecting an integral type
```

It seems that there are problems dealing with `yarp::conf::createVocab32()`.

cc @traversaro 

### Steps to reproduce

Link to failed CI:
- https://github.com/robotology/icub-main/actions/runs/11321719276/job/31481204884

traversaro, 2024-10-14:
I guess `ubuntu-latest` switched from being 22.04 to 24.04 . Ubuntu 24.04 comes with swig 4.2.0, that has some known bugs that are solved in later version of 4.2.1 , for that we are installing a custom (private, so it does not interfere with the rest of the system) swig in the robotology-superbuild when building on 24.04 .

The quickest fix is to go back to 22.04, otherwise to test with 24.04 with apt dependencies we need to manually install a newer swig version.

pattacini, 2024-10-14:
Thanks @traversaro 👍🏻 

Here's for reference the superbuild machinery you mentioned:
- https://github.com/robotology/robotology-superbuild/blob/master/cmake/RobotologySuperbuildLogic.cmake#L39-L64

If updating via apt is quick, I'll do that, otherwise I'll switch it back to 22.04.

pattacini, 2024-12-09:
✨✨ Here's [an AI-assisted sketch of how you might approach this issue](https://copilot-workspace.githubnext.com/robotology/icub-main/issues/988?shareId=29676f31-1a31-49f6-a0b5-11c642f5912a) saved by @pattacini using Copilot Workspace v0.27

commit 61a66a35 (2024-12-09): Fix CI failure with SWIG::Python on Ubuntu 24.04

Fixes #988

Update CI configuration to use Ubuntu 24.04 and custom SWIG 4.2.1

* Change the CI configuration in `.github/workflows/ci.yml` to use `ubuntu-24.04` instead of `ubuntu-22.04`.
* Add steps to install a custom build of SWIG 4.2.1 for Ubuntu 24.04.
* Remove the step to install SWIG via `sudo apt install`.
* Ensure the custom SWIG 4.2.1 is used for the build process on Ubuntu 24.04.

---

For more details, open the [Copilot Workspace session](https://copilot-workspace.githubnext.com/robotology/icub-main/issues/988?shareId=XXXX-XXXX-XXXX [... cut]
PR #999 (2024-12-09): Fix CI failure with SWIG::Python on Ubuntu 24.04
Fixes #988

Update CI configuration to use Ubuntu 24.04 and custom SWIG 4.2.1

* Change the CI configuration in `.github/workflows/ci.yml` to use `ubuntu-24.04` instead of `ubuntu-22.04`.
* Add steps to install a custom build of SWIG 4.2.1 for Ubuntu 24.04.
* Remove the step to install SWIG via `sudo apt install`.
* Ensure the custom SWIG 4.2.1 is used for the build process on Ubuntu 24.04.

---

For more details, open the [Copilot Workspace session](https://copilot-workspace.githubnext.com/robotology/icub-main/pull/999?shareId=a324a996-0248-44c [... cut]
closed 2024-12-09 by pattacini