robotology/icub-main #991: Cleanup workaround in rawValues devices param parsers

Nicogene, 2024-11-13:
In:
- #983 

the `rawValues` debug sw pipeline has been added (interfaces, nwc, nws) to icub-main, and the devices exploit the new feature of param parsing generation of YARP.

Since YARP was not released yet @MSECode added manually the header needed, but since that now YARP has been released:

- https://github.com/robotology/yarp/releases/tag/v3.10.0

This workaround can be removed, and icub-main can require `3.10.0` as yarp required version.

cc @valegagge

MSECode, 2025-01-08:
Cleaned up the workaround with this commit: https://github.com/MSECode/icub-main/commit/208ad3934386774df15bae1cf16a89c20a1ccd93

Ready for the PR. To clean old modified files I'll pushed just this time the generated files.

MSECode, 2025-01-13:
Hi @Nicogene,
should I enable the plugins of the `rawValuesPublisher` by default or should I keep them OFF when building icub-main by default, what do you think?

pattacini, 2025-01-13:
Hi @MSECode 

Turn it `OFF`, please.

closed 2025-01-13 by pattacini

traversaro, 2025-01-15:
Should we enable them somewhere, for example in a preset ( https://github.com/robotology/icub-main/blob/master/CMakePresets.json#L23 )? Otherwise, how can users (even internal to IIT, that will be the main target of this feature) use this interface?

Other places were we may want to enable them (that I would like to substitute with preset at some point):
* superbuild icub recipe: https://github.com/robotology/robotology-superbuild/blob/6f38a8c262d332cf3e9844cf6455729f16add446/cmake/BuildICUB.cmake#L68
* conda icub recipe: https://github.com/conda-forge/icub-main-feedstock/blob/107574db70dd2317711a62712d39f26436342721/recipe/build_cxx.sh#L24 and https://github.com/conda-forge/icub-main-feedstock/blob/107574db70dd2317711a62712d39f26436342721/recipe/bld_cxx.bat#L6

pattacini, 2025-01-15:
My idea was to handle this module like the Cartesian and the Gaze components, which should be OFF by default.

Now, if it's a matter of exposing only the I/F in any condition we could do that but we should keep the building of the relative implementation plus the fake stuff still OFF.

A shortcut could be relying on preset anyway as you suggested.

traversaro, 2025-01-15:
> My idea was to handle this module like the Cartesian and the Gaze components, which should be OFF by default.

I totally agree, that make sense, but indeed then we enable the cartesian and gaze devices everywhere else, including in the `default` preset. I guess @MSECode has an idea on how these devices will be used by users, and can suggest which options needs to be enabled (at least in the superbuild).

MSECode, 2025-01-17:
Yes leaving them off it is totally fine since that module should mainly have debug purposes thus it is fine that if someone needs it it will enable and use it.

traversaro, 2025-01-17:
> Yes leaving them off it is totally fine since that module should mainly have debug purposes thus it is fine that if someone needs it it will enable and use it.

At least related to the superbuild, based on my experience I am not sure if this is a good idea. Enabling one (or several) CMake options in a subproject of the superbuild is not an intuitive (and/or fast operation) for most users, and if you are using Docker (depending on how the image is structure) it may require a long rebuild of the full image, or it simply impossible if you are using conda binaries. Documenting a long list of CMake options to enable and/or disable depending on what a user needed to do was what we did in the past (https://icub-tech-iit.github.io/documentation/sw_installation/icub_head_manual/) and it was always a source of confusion as people forgot or did errors in setting the required options. Can you at least document the options that are required to enable this functionality somewhere so that we can independently enable these options in the superbuild?

MSECode, 2025-02-10:
Hi @Nicogene, @traversaro,
other then https://icub-tech-iit.github.io/documentation/sw_installation/icub_head_manual/ where could I add a few lines for document how to enable the `rawValuesPublisher` since that manual seems quite outdated now. Should we add it directly in the icubmain docs?

pattacini, 2025-02-10:
> other then https://icub-tech-iit.github.io/documentation/sw_installation/icub_head_manual/ where could I add a few lines for document how to enable the `rawValuesPublisher` since that manual seems quite outdated now.

Actually, it's not outdated. It has been refreshed via https://github.com/icub-tech-iit/documentation/pull/380 because of the use of presets.

The use of presets was an option as we said above.
Another option is to enable the device in the superbuild.

traversaro, 2025-02-10:
To be honest, I am not sure and that is exactly the reason why I suggested enabling the options in the superbuild, so that everything works out of the box without hoping that users read some documentation. Anyhow, an initial (even if incomplete, as users are not expected to read issue) form of documentation is start writing down the option that need to be enabled in the relevant issue or PRs.

pattacini, 2025-02-10:
> Oks, understood, thus for now I can add some documentation there. Then we can see if we can enable it by default.

Nope, we won't enable it by default in icub-main, as we said, unless you want to update the [presets](https://github.com/robotology/icub-main/blob/master/CMakePresets.json). Updating the presets is sufficient, thus there's no need for updating the documentation.

That said, the best option still remains to update the superbuild as @traversaro suggested earlier.
You can drop here the relative CMake flags for that.