robotology/icub-main #992: CI Python bindings failed w/ `yarp-3.10.0`

pattacini, 2024-11-13:
### Bug description

Here's the CI error:

```console
2024-11-13T12:05:05.4061826Z /home/runner/work/icub-main/icub-main/install/share/yarp/bindings/yarp.i:437: Warning 206: Unexpected tokens after #endif directive.
2024-11-13T12:05:05.4084387Z /home/runner/work/icub-main/icub-main/install/share/yarp/bindings/yarp.i:2013: Error: Unable to find 'swig_python_windows_preable.i'
2024-11-13T12:05:05.4205031Z gmake[2]: *** [bindings/CMakeFiles/icub_python_swig_compilation.dir/build.make:76: bindings/CMakeFiles/icub_python.dir/icubPYTHON.stamp] Error 1
2024-11-13T12:05:05.4206427Z gmake[2]: *** Deleting file 'bindings/CMakeFiles/icub_python.dir/icubPYTHON.stamp'
2024-11-13T12:05:05.4207733Z gmake[1]: *** [CMakeFiles/Makefile2:4548: bindings/CMakeFiles/icub_python_swig_compilation.dir/all] Error 2
2024-11-13T12:05:05.4210833Z gmake: *** [Makefile:146: all] Error 2
2024-11-13T12:05:05.4237755Z ##[error]Process completed with exit code 2.
```

### Steps to reproduce

Failed CI:
- https://github.com/robotology/icub-main/actions/runs/11816589875

traversaro, 2024-11-13:
Fix provided in https://github.com/robotology/yarp/pull/3148, it is a regression of https://github.com/robotology/yarp/pull/3146 .

pattacini, 2024-11-26:
[YARP-3.10.1](https://github.com/robotology/yarp/releases/tag/v3.10.1) has been released and now the CI runs happily. See for example:
- https://github.com/robotology/icub-main/actions/runs/12038477411.

Closing.

closed 2024-11-26 by pattacini