robotology/icub-main #993: ARE module failed to run with Gazebo for iCub

omareldardear, 2024-11-25:
### Bug description

I am trying to launch actionRenderingEngine module with Gazebo. 
I get the following error: 
......
[ERROR] |yarp.device.multipleanalogsensorsclient| Failure connecting port /actionsRenderingEngine/springy/left_hand/MAIS/rpc:i to /icubSim/left_hand/MAIS/rpc:o.
[ERROR] |yarp.device.multipleanalogsensorsclient| Check that the specified MultipleAnalogSensorsServer is up.
[ERROR] |yarp.dev.PolyDriver|multipleanalogsensorsclient| Driver <multipleanalogsensorsclient> was found but could not open
[ERROR] *** actionsRenderingEngine/springy: unable to connect to /icubSim/left_hand/MAIS
.....
[ERROR] |yarp.os.impl.ThreadImpl| A thread failed to start with error code: 0
[ERROR] ARE could not be initialized!

I use the standard process following the main template 

### Steps to reproduce

OS: Ubuntu 20.04 
Superbuild: v2024.08.0 
Gazebo 11
Using the following process

* $ yarpserver
* $ gazebo 
* Placing iCub Model -> iCubGazeboV2_5 (no_hands) 
* $yarprobotinterface --context gazeboCartesianControl --config no_legs.xml
* $ iKinCartesianSolver --context gazeboCartesianControl --part right_arm
* $ iKinCartesianSolver --context gazeboCartesianControl --part left_arm
* $ iKinGazeCtrl --context gazeboCartesianControl --from iKinGazeCtrl.ini
* $wholeBodyDynamics --robot icubSim --autoconnect --dummy_ft --headV2 --no_legs
* $actionsRenderingEngine --from gazebo/config.ini

omareldardear, 2024-11-25:
Hi @Nicogene , 

Thank you for your reply. 
Indeed, I tried with visuomanip model too and I got the same issue. 
I will wait for the new release then.

pattacini, 2024-12-09:
Hi @omareldardear 

Distro 2024.11.1 is available for you to test:
- https://github.com/orgs/robotology/discussions/720

omareldardear, 2024-12-09:
Great! 
I just tested ARE with Gazebo and iCub in a docker container and everything seems to work nicely. 

Many Thanks :)

closed 2024-12-09 by omareldardear