robotology/icub-main #1000: Improve the diagnostic for temperature values and remove calib14 debug lines

MSECode, 2024-12-11:
### Is your feature request related to a problem?

As highlighted in here: https://github.com/robotology/icub-tech-support/issues/1987#issue-2720817331, the diagnostic message for the temperature errors need to be revised as follows:
- timestamp should be formatted to a readable relative time or, if we wanna keep it absolute, with readable time and data and not as just a number in milliseconds.
- we can think to move the errors to warnings since they are non blocking and they do not move the joint in fault.

Moreover, we need to remove the following debug lines from the calibration 14 logs, since we do not need them anymore:
https://github.com/robotology/icub-main/blob/22c06ceffe66c21de446da841a1ffd432ad515d0/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L2285-L2290

### The solution you would like to have available

Improve diagnostic by:
- formatting well the timestamp to a readable string for the temperature related issue
- add the converted temperature in celsius, so that the user knows the actual value of the raw temperature sent in the logs
- remove the unneeded logs from the calibration 14

MSECode, 2024-12-13:
Hi @valegagge,
ragarding timestamp and debug string all ok.
Considering instead the issue with the conversion of the raw temperature to Celsius at this section of the parser: https://github.com/robotology/icub-main/blob/22c06ceffe66c21de446da841a1ffd432ad515d0/src/libraries/icubmod/embObjLib/diagnosticInfoParsers.cpp#L704-L715,
we would make use of the methods of the interface: `yarp::dev::eomc::ITemperatureSensor`, which have already been overridden in here: https://github.com/robotology/icub-main/blob/22c06ceffe66c21de446da841a1ffd432ad515d0/src/libraries/icubmod/embObjMotionControl/eomcParser.h#L79-L233, depending on the type of sensor.
Thus, to exploit them, I was thinking to add in the constructor of `Diagnostic::LowLevel::MotionControlParser` a unique_ptr to each of the override of  `yarp::dev::eomc::ITemperatureSensor` thus to use the method: `virtual double convertRawToTempCelsius(const double temperature)`

Does it make sense? Can this lead to problem of circular dependencies considering that the diagnostic is part of `embObjLib` and `eomcParser` is part of `embObjMotionControl`? Is there a better manner to do this without the need of re-overriding the base interface? 
We can discuss about that the next days.

valegagge, 2024-12-16:
@MSECode , thanks. It is clear. We are going to talk as soon as possible!

MSECode, 2024-12-24:
After discussion we proceed in fixed the following bugs:
- timestamps error for the temperatures errors and set them as warnings. Here the error: https://github.com/robotology/icub-tech-support/issues/1987#issuecomment-2521270283
- removal of debug lines for calibration type 14

Regarding the conversion of the temperatures from raw to Celsius on the errors will be addressed later since it cannot be currently done without dealing with circular dependencies on icub-main.

closed 2025-01-08 by pattacini