robotology/icub-main #1005: Improve the motor temperature diagnostic

valegagge, 2025-01-13:
### Is your feature request related to a problem?

As explained in https://github.com/robotology/icub-tech-support/issues/2007, when a motor sensor temperature doesn't read correctly the diagnostic reports the `overheating` error. 
In the first development of this feature, we used the same error code `overheating` both for the real `overheating` and also for error readings. The user can discriminate between the two errors checking the raw value of temperature. if is is less than zero means an reading error occurred.

In the ticker mentioned above we noticed that, only the first diagnostic error report the negative value, while the next one report the `overheating` error with a positive value making the user confused, if it is a fake overheating.

So, we need to avoid this misunderstanding.

### The solution you would like to have available

I think the best solution is adding a different error type in the fw and then improving the diagnostic.

valegagge, 2025-04-07:
Hi @MSECode , please add a comment here.

MSECode, 2025-04-07:
This has been solved and completed. Details can be browsed in the comments of this PR: https://github.com/robotology/icub-firmware/pull/580 where the updates to the 2FOC code have been completed.
Now, if we have a i2c communication error on the TDB this is shown to the user with the correct message: `MC: TDB cannot receive any data on I2C channel. Motor temperature detection board is not receiving any data on the i2C channel.", "In par16 = ID of the joint.`

closed 2025-04-07 by MSECode