robotology/icub-main #1016: Improve the calibration of joint with POS service

valegagge, 2025-04-15:
### Is your feature request related to a problem?

Often, the ergocub users experienced issues or difficulties in calibrating the joint with POS services.
This issue aims to make this procedure straightforward and more transparent. 

### The solution you would like to have available

Remove the calibration data from the POS service configuration file, leaving the calibration info in the calibration files. 

If necessary, improve the documentation for the calibration based on POS. (`calibration 14`)

MSECode, 2025-04-15:
As you can see from these lines: https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjLib/serviceParser.cpp#L884-L927 even if we remove the configuration values for the POS calibration section default values will be set. Thus we won't get an error. Moreover, if we use the calibration type 14 for all the joints that enable POS service the calibration data can be passed from the calibration file and fed first to the motion control, which then updates the struct of the POS service. Thus, I would first suggest to remove the POS calibration section from the configuration files, calibrate all fingers with calibration type 14, thus also the abduction, and see the results.

cc: @valegagge

valegagge, 2025-04-16:
For the sake of completeness, please verify on the  robot the following points:
- what happen if we start the robot without the pos service? What does the user see as encoder value in motorigui?
- if skipCalibration is true, what does the user see as encoder value in motorigui?

In addition, in order to avoid the user adding calibration parameters in the POS service, I suggest adding a check in the parser that signals that such calibration parameters are ignored by an error message and failing to open the POS service. What do you think?

MSECode, 2025-04-16:
# Observations

## Ports that stream arm values

- I've noticed that at the moment the POS folder wrapper are not written inside the robot xml file in the configuration files. As a matter of fact the ports `/robotname/*_hand/POS2:o` and `/robotname/*_hand/POS4:o` are not open among the all ports that we can see when asking `yarp name list`. We just configure the files for the hw part, i.e. `hardware/POS/*_hand-pos*.xml` 
-

MSECode, 2025-05-02:
## Current status and decisions about interface and configuration

After some tests done on the robot and after study of `embObj*` code related to the POS service we have come up with the following observations:
- first of all if we either remove or keep the `CALIBRATION` group in `motionControl` part of the configuration files, e.g. https://github.com/robotology/robots-configuration/blob/master/ergoCubSN001/hardware/motorControl/left_arm-eb23-j7_10-mc_service.xml#L42-L48, nothing changes. This is due to 2 reasons: 
  - one is that the mc data related to the POS service are set to zero and thus, that part of the configuration files is never parsed (check here: https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjLib/serviceParser.cpp#L4327-L4336)
  - second, that need to be checked to be 100% sure, we are quite confident that those objects that are set to the default values are not even sent to the boards. Therefore, it is only the `embObjPOS` service that consider that configuration group not the motion control service

- instead if we currently remove the `CALIBRATION` group from the `POS` folder configuration files we cannot calibrate the abduction joint (only index and thumb abductions). It should be noted that all the open-close are instead correctly calibrated even if we remove the group. 
- our decision for proceeding is the following: remove from all the configuration files the `CALIBRATION` group, both from the MC and POS folders and add to the code of the `serviceParser` an error that won't make you proceed if that group is found in the `POS` folder configuration files. Then, in order to correctly calibrate the abduction we need to use the `calibration type 14` for those. Thus, in relation to this, I'd like to know were I can find a diagram or something else that show me the how the kinematics of the robot hands is defined. Specifically, I'm interested in knowing where should stay the thumb and index abduction at zero and at 90 degrees. (Maybe @AntonioConsilvio knows this)

Updates to the icub-main code are available here: https://github.com/MSECode/icub-main/tree/feature/posUpdate
As soon as we completely tested them and we calibrated the abduction with the new calibration we are going to update robots-configuration as well.

pattacini, 2025-05-02:
Thanks @MSECode for the update 👍🏻 

CC'ing some other people in the know: @marcoaccame @Nicogene.

> Thus, in relation to this, I'd like to know were I can find a diagram or something else that show me the how the kinematics of the robot hands is defined. Specifically, I'm interested in knowing [...]

Some related resources are available in our [documentation](https://icub-tech-iit.github.io/documentation/hands/). At any rate, I would also recommend that you reach out to our mech designers as well for a double-check, cc @fiorisi @lawproto.

Given the expected quite big impact to the robot configuration files, it's paramount that we deal with that with a proper announcement that can effectively land in the upcoming PR on robots-configuration repo.

Lawproto, 2025-05-06:
> Thus, in relation to this, I'd like to know were I can find a diagram or something else that show me the how the kinematics of the robot hands is defined. Specifically, I'm interested in knowing [...]

Hope these images can help clear any doubt:

[image]

[image]

[image]

[image]

When we established which position should be considered zero in the CAD I believed we were following some kind of software guidelines, however this request makes me believe it is not so, after all.

MSECode, 2025-05-06:
Following those standard and the documentation we have updated the robot hand calibration as shown in this commit: https://github.com/icub-tech-iit/robots-configuration/commit/16cbc1eb1963ff9566ad0141d0fc42f9495896bb. Specifically, we have done the following:
- re-enabled left and right thumb abduction and calibrate them with calibration type 14
- enabled only right arm index abduction since left arm index abduction has the FAP encoder that is probably detached or broken since it is not reading any value and calibrate the working joint with calibration type 14
- removed the CALIBRATION group from POS folders since it is not needed when using calibration type 14, whose calibration parameters are managed by the MC, which then send the calibration information to the POSService.

## Some NOTES:
When updating the calibration values I've noticed couple of things that should be better investigated:
- the zero position raw offset tends to change after time (probably due to errors with the FAP sensor which is magnetic and thus might have some issues over time)
- the joint soft limits tends to change over time --> I needed to reduce a lot the range of motion especially for the right arm --> I suppose this is mostly due to the cables that loose elasticity over time, thus when the joint motor arrive to the hard stop the joint is not at its full stop (ideally 90 degrees in one direction and zero towards the other)
- PIDs for abduction needs to be adjusted since they move quite slowly even with 100 degrees/s in velocity 

Anyway I tested on `ergoCub001` the updated calibration values and the robot arms calibrate fine. We can then plan to investigate the point left above.
I'll close this issue and eventually we can open another for the work to be done defined above.

cc:: @valegagge

pattacini, 2025-05-06:
>Following those standard and the documentation we have updated the robot hand calibration as shown in this commit: https://github.com/icub-tech-iit/robots-configuration/commit/16cbc1eb1963ff9566ad0141d0fc42f9495896bb.

cc @S-Dafarra @PasMarra